PR2/pr2_ethercat_drivers #68: Request of indigo release

garaemon, 2015-01-12:
Could you release deb of this package on indigo?

TheDash, 2015-01-12:
I can try, but it has lots of dependencies, so we will see if that is possible at this exact moment.

garaemon, 2015-01-12:
thanks! I think eml package is not available on indigo...

TheDash, 2015-01-12:
I don't think eml is required. It used to be, but does it provide any value?

TheDash, 2015-01-12:
Requires pr2_common, I've included eml in the indigo release.

trainman419, 2015-01-15:
EML is the EtherCat Master Library - definitely required as a dependency for releasing the pr2 ethercat drivers.

The ROS package was maintained as part of the PR2 software in the past; you should have release rights for it. If you can't figure out how to release it, let me know and I'll take a deeper look.

TheDash, 2015-01-15:
It was on the groovy branch on this repo and I catkinized it and released it into indigo. I don't believe the deb built tho.. So I'll see what's up with it.

trainman419, 2015-01-15:
EML shouldn't be catkinized; it should be released as a third-party package. That's why it was removed from the hydro branch.

TheDash, 2015-01-15:
Ohh I see what you mean now. Theres the eml package with the patch, but he actual .so needs to be released too, yeah?

trainman419, 2015-01-15:
The release repo for EML is: https://github.com/ros-gbp/eml-release

The [tracks.yaml](https://github.com/ros-gbp/eml-release/blob/master/tracks.yaml) specifies that the source should be downloaded from pr.willowgarage.com, patched with the included package.xml, and then built into a deb on the farm.

Since pr.willowgarage.com no longer exists, I've uploaded the source tarball to my server: http://namniart.com/eml-r36.tar.gz . Please host it somewhere else and update the tacks.yaml in release repo accordingly.

Once you have the source hosted in a new location, and the tracks.yaml updated, you should re-release the current version in hydro; confirm that it works, and then create a new release track for the indigo release and run the bloom release for indigo.

trainman419, 2015-01-15:
@TheDash really? you forked the groovy-devel branch to create the Indigo release, and potentially discarded all of the patches and fixes that went into Hydro? that's a great way to ruin the version history of a repository and make it very difficult to keep track of changes.

I suspect the changes  between Hydro and Indigo are so small that you don't even need to maintain a separate indigo-devel branch, assuming that you actually release the eml package properly.

In the future, you should check the debbuild page and the rosdistro index before assuming that a package was removed for Hydro; there were a significant number of PR2-related packages that I migrated to third-party packages for the Hydro release.

TheDash, 2015-01-15:
My main intent with that was to get the other pr2 packages released into the distribution yaml, some others had been requiring it for their releases, and I hadn't really looked into whether the contents were achieving anything yet. That was the quickest way to do it with the resources I had. Pretty sure I didn't understand what was going on with the eml situation.

Are you saying that the indigo devel branch is unnecessary? I'm convinced that its convention to create a source branch w.r.t the rosversion its for.

I understand what you mean now. Eml was supposed to be released as third party but is now released under ethercat drivers. 

Is this going to cause issues with indigo, or can it be fixed by pointing the source of eml to the new location?

TheDash, 2015-01-15:
Oh wait what, I see what you mean by skipping hydro. I am pretty sure users can just look at the hydro branch to see the commits made from there. The changes aren't lost. They just aren't merged into one stream.

trainman419, 2015-01-16:
EML was an old, third-party package which built primarily using special rosbuild Makefile targets; not the CMakeLists (go read the Makefile). These targets aren't supported by catkin, and is why I converted it into a third-party package.

You should delete the indigo-devel branch, remove the releases you've made from rosdistro, and start over. Release the third-party version of EML into Indigo, verify that it builds, and then install the eml deb and verify that the hydro-devel branch builds on indigo. Once you've confirmed that, _then_ you can release the hydro-devel branch into Indigo.

TheDash, 2015-01-16:
Gotcha. EML first, and then pr2_ethercat_drivers.

garaemon, 2015-02-01:
http://jenkins.ros.org/view/IbinS32/job/ros-indigo-eml_binarydeb_saucy_i386/20/console

Still failing to build EML. any update?

TheDash, 2015-02-01:
That's right, I haven't had a chance to do the fix Austin mentioned. Tomorrow though, I'll be scheduled to get it started. We've just received a new set of servers from WG, and I was trying to put 14.04 on them in the past week.

TheDash, 2015-02-03:
Thoughts on this @ahendrix? https://github.com/shadow-robot/ros_ethercat

closed 2015-06-19 by TheDash